smartinick/esp32_xiaomi_m365 #6: Not Receiving ESC Packets, and BMS packets.

xinux87, 2018-07-31:
HI!!
Now i have all connected but i'm not getting any ESC or BMS Packets.

I thing that there is some kind of mistake, i have esp32-hal-uart.c patched.

Thanks!

RAW telnet SERVER:
[image]

Data/Telemetry Server packet overview
[image]

smartinick, 2018-07-31:
OK, let's start this step by step... 

1st: if your flashed your esp32 using OTA, then just reset it once. something is messed up also with my esp32, after flashing via OTA, it doesn't receive data...

2nd: _if_ you did all the changed to the esp32_hal_uart.c this problem-source should be obosolete. to go shure, you can look into my github account, i've forked the esp32-arduino source and compare your esp32-hal-uart with the one in my repo. (or compare with base and see the changes/reapply them).

3rd -> disconnect the TX from ESP, so you can only receive data (OR: Comment out the call to m365_handlerequests(); in void m365_receiver.
then you should not request/get bms/esc packets... not if you start a app (ninebot, mihome) and read data from the scooter, it should show you data on the telnet screens... if not - there's a error with the connections or gpio pins. but as you receive something, i guess it's _not_ a wiring problem.
but that would be the basic check to be sure!

4th - next step would be to send requests (connect TX or uncomment the previously commented m365_handlerequests())

and i hope that we have a hit in those 4 steps.... ;)

xinux87, 2018-07-31:
1st of all thanks! for your fast response.

1st: i'm debuggind directy using USB port
2nd: cheked and changed to use your fork. (same behavior)
3rd: commented m365_handlerequest, the Raw log dont have the big sentences with crc error.
and if y connect with an APP, i receive good sentences. (i receive good sentences ESC and BMS if i leave the hadlerequest active.

Humm... something missing with the board? i'm using esp32 generic board...

Thanks!!!

smartinick, 2018-08-01:
OK, so we know it's not a problem with receiving data, but with sending the requests (bms/esc do not send data on their own, we (app or µC) have to request the data.
So there's a problem with your TX line,maybe with the diode...?

i'll push a pic of the schematics to the repo soon, forget the oled-stuff there - it's not tested. but you can see the 2 R and the Diode as i use it.

and i guess you are using the correct GPIO pins...?  on esp32 you  should be using those defined via UART2RX, UART2TX and the GPIO defined in UART2RXunused should not be connected to anything...

by the way.... the CRC errors as shown in your screenshot are "just" 2% which is not a problem. i've not seen problems with driving with 30% crc errors, though this is a issue on my todolist - to get it to 0%.

xinux87, 2018-08-01:
Solved!! 
I was connecting the resistor beafore the diode. thanks to the schematic!!! and the values.
there ir a photo: with values and diode orientation, for newbies:

[image]

Thanks!!!!

closed 2018-08-01 by xinux87