smartinick/esp32_xiaomi_m365 #8: ESP32 single initial screen in loop

Pi3t, 2018-08-06:
Hello, I have programmed your code in an esp32 and I only get the initial screen with the drawing and after a few seconds it restarts, you could help me, thank you in advance and congratulations for your great work. a greeting

smartinick, 2018-08-07:
well brownout means there's a tooo low Vcc, and disabling the brownout detector will not solve that problem.
so there's some problem with your gnd/5v/3.3v cabling/voltage regulators... is gnd wired tom365 gnd?
enable the brownout detector and solve the problem... :D
is 5v wired to m365 5v? how do you create the 3.3v? can the 3.3v voltage regulator provide at least 300-350mA?
do you have the typical capacitors in front/behind the voltage regulators, the typical 100nF at the esp32 gnd/vcc pins..?

smartinick, 2018-08-29:
closed due to no feedback

closed 2018-08-29 by smartinick