smartinick/esp32_xiaomi_m365 #9: [feature request] option to configure wheel size

ferboiar, 2018-08-16:
m365 calculates the speed depending on the size of the wheel, default for 8.5 ". 

If you mount 10" wheels the speed and distance traveled shown will not be correct. It is necessary to multiply the displayed value by a correction factor. It would be nice an option to choose the size of the installed wheel so the displayed data would be real.

S-m-i-t-h, 2018-08-29:
do you know more or less when this will be released?
it will be included in a configuration menu?

smartinick, 2018-08-29:
Just added it.
for now it's a #define wheel10inch which you have to comment/uncomment.... if you look for #ifdef wheel10inch you see that i define a factor which is 1 or 10/8.5.

it compiles, i didn't actually flash it or test it - so please verify yourself and give me feedback - Thanks!

I'm using it only on the oled-display routines (telnet-screen still shows esc-reported values) and i apply the factor to speed, average speed and tripdistance.
it's not applied to totaldistance, which could make sense if you buy a new scooter and start with 10" wheels.

and i do not apply it to "remaining distance" - the estimation how many km we will ride with this battery...  for this point - any idea...? it would be interesting - in your driving style:how many kilometers do you make on a charge with 8.5" and how many with 10"...? or e.g. on your daily office-ride - how much batt-percent on average do you have with8.5" and how much with 10" left after arriving. maybe we can get any idea there.... i'd believe it's no difference.... (less rpm but more power/current needed and in the end it's the same air-resistance,...)

and once i've made the config menu, there will be a option to change that on the scooter as well and not only via a #define/recompile/reflash.

closed 2018-09-02 by smartinick