smartinick/esp32_xiaomi_m365 #11: OTA by wifi using web browser

S-m-i-t-h, 2018-08-23:
Hi,
I change settings (SSID & Wifi password) to my own wifi and ESP32 connected succesfully to the WIFI.

Using the web browser I haver tried to go to the assignated IP, but I get on the web browser "ERR_EMPTY_RESPONSE"

I tryied also using " http://esp32.local" or " http://ESP32.local"

smartinick, 2018-08-23:
Currently i've only implemented the OTA version that runs from within the ide. So you can compile&upload from arduinoide and others supporting that method.

Http-upload is more code and more steps necersarry to actually do the update... I don't see the sense in that?

S-m-i-t-h, 2018-08-23:
OK, understood. I have tried and it works fine. 

I think that the http upload is interesting for people that want to upload the .bin file without installing arduino IDE and without compiling again the source file before uploading. (i mean in a easier way). 

Thanks for your time!

closed 2018-08-25 by S-m-i-t-h

smartinick, 2018-08-29:
you might be right, but how provides them the .bin? :D

once i add a http-interface, i'll add that option as well... right now i've still some basics to implement :D