smartinick/esp32_xiaomi_m365 #12: Compatibility with Arduino

augisbud, 2018-08-27:
Hi so you say that it's compatible with arduino however it doesn't compile???

smartinick, 2018-08-28:
Hi, compiles...

`esptool.py v2.3.1

Multiple libraries were found for "WiFi.h"
 Used: c:\...\Documents\Arduino\hardware\espressif\esp32\libraries\WiFi
 Not used: c:\...\programming\arduino-1.8.5\libraries\WiFi
Using library WiFi at version 1.0 in folder: c:\...\Documents\Arduino\hardware\espressif\esp32\libraries\WiFi 
Using library Update at version 1.0 in folder: c:\...\Documents\Arduino\hardware\espressif\esp32\libraries\Update 
Using library Preferences at version 1.0 in folder: c:\...\Documents\Arduino\hardware\espressif\esp32\libraries\Preferences 
Using library ArduinoOTA at version 1.0 in folder: c:\...\Documents\Arduino\hardware\espressif\esp32\libraries\ArduinoOTA 
Using library git_smf_Adafruit_SSD1306 at version 1.1.2 in folder: c:\...\Documents\Arduino\libraries\git_smf_Adafruit_SSD1306 
Using library SPI at version 1.0 in folder: c:\...\Documents\Arduino\hardware\espressif\esp32\libraries\SPI 
Using library Adafruit_GFX_Library at version 1.2.3 in folder: c:\...\Documents\Arduino\libraries\Adafruit_GFX_Library 
Using library ESPmDNS at version 1.0 in folder: c:\...\Documents\Arduino\hardware\espressif\esp32\libraries\ESPmDNS 
Using library Wire at version 1.0 in folder: c:\...\Documents\Arduino\hardware\espressif\esp32\libraries\Wire 
Sketch uses 761606 bytes (58%) of program storage space. Maximum is 1310720 bytes.
Global variables use 51524 bytes (15%) of dynamic memory, leaving 276156 bytes for local variables. Maximum is 327680 bytes.
`

maybe you can provide some more information?

smartinick, 2018-08-28:
And for your convenience.... also on platformio there's no problem with compiling:

Processing esp32dev (platform: espressif32; lib_extra_dirs: c:\...\Documents\Arduino\libraries; board: esp32dev; upload_flags: --auth=ldhgfjdlkg; framework: arduino)

PLATFORM: Espressif 32 > Espressif ESP32 Dev Module
SYSTEM: ESP32 240MHz 288KB RAM (1.25MB Flash)
xtensa-esp32-elf-g++ -o "c:\...\programming\arduino_github_esp\esp32_xiaomi_m365\esp32_m365_oled\esp32_m365_oled.ino.cpp" -x c++ -fpreprocessed -dD -E "c:\...\appdata\local\temp\tmpfct7nt"
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 66 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <Update> 1.0 (c:\...\.platformio\packages\framework-arduinoespressif32\libraries\Update)
|-- <ArduinoOTA> 1.0 (c:\...\.platformio\packages\framework-arduinoespressif32\libraries\ArduinoOTA)
|   |-- <WiFi> 1.0 (c:\...\.platformio\packages\framework-arduinoespressif32\libraries\WiFi)
|   |-- <Update> 1.0 (c:\...\.platformio\packages\framework-arduinoespressif32\libraries\Update)
|   |-- <ESPmDNS> 1.0 (c:\...\.platformio\packages\framework-arduinoespressif32\libraries\ESPmDNS)
|   |   |-- <WiFi> 1.0 (c:\...\.platformio\packages\framework-arduinoespressif32\libraries\WiFi)
|-- <PubSubClient> 2.6 (c:\...\Documents\Arduino\libraries\pubsubclient-master)
|-- <WiFi> 1.0 (c:\...\.platformio\packages\framework-arduinoespressif32\libraries\WiFi)
|-- <Adafruit SSD1306> 1.1.2 (c:\...\Documents\Arduino\libraries\git_smf_Adafruit_SSD1306)
|   |-- <Wire> 1.0 (c:\...\.platformio\packages\framework-arduinoespressif32\libraries\Wire)
|   |-- <Adafruit GFX Library> 1.2.3 (c:\...\Documents\Arduino\libraries\Adafruit_GFX_Library)
|   |   |-- <SPI> 1.0 (c:\...\.platformio\packages\framework-arduinoespressif32\libraries\SPI)
|   |-- <SPI> 1.0 (c:\...\.platformio\packages\framework-arduinoespressif32\libraries\SPI)
|-- <Preferences> 1.0 (c:\...\.platformio\packages\framework-arduinoespressif32\libraries\Preferences)
|-- <Adafruit GFX Library> 1.2.3 (c:\...\Documents\Arduino\libraries\Adafruit_GFX_Library)
|   |-- <SPI> 1.0 (c:\...\.platformio\packages\framework-arduinoespressif32\libraries\SPI)
xtensa-esp32-elf-g++ -o .pioenvs\esp32dev\src\esp32_m365_oled.ino.cpp.o -c -fno-rtti -fno-exceptions -std=gnu++11 -Wno-error=deprecated-declarations -Wno-error=unused-function -Wno-unused-parameter -Wno-sign-compare -fstack-protector -Os -g3 -nostdlib -Wpointer-arith -Wno-error=unused-but-set-variable -Wno-error=unused-variable -mlongcalls -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -DARDUINO=10805 -DARDUINO_ARCH_ESP32 -DARDUINO_VARIANT=\"esp32\" "-DARDUINO_BOARD=\"Espressif ESP32 Dev Module\"" -DESP32 -DESP_PLATFORM -DF_CPU=240000000L -DHAVE_CONFIG_H -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -DPLATFORMIO=30504 -DARDUINO_ESP32_DEV -Ic:\...\programming\arduino_github_esp\esp32_xiaomi_m365\esp32_m365_oled -Ic:\...\.platformio\packages\framework-arduinoespressif32\libraries\Update\src -Ic:\...\.platformio\packages\framework-arduinoespressif32\libraries\WiFi\src -Ic:\...\.platformio\packages\framework-arduinoespressif32\libraries\ESPmDNS\src -Ic:\...\.platformio\packages\framework-arduinoespressif32\libraries\ArduinoOTA\src -Ic:\...\Documents\Arduino\libraries\pubsubclient-master\src -Ic:\...\.platformio\packages\framework-arduinoespressif32\libraries\Wire\src -Ic:\...\.platformio\packages\framework-arduinoespressif32\libraries\SPI\src -Ic:\...\Documents\Arduino\libraries\Adafruit_GFX_Library -Ic:\...\Documents\Arduino\libraries\git_smf_Adafruit_SSD1306 -Ic:\...\.platformio\packages\framework-arduinoespressif32\libraries\Preferences\src -Ic:\...\.platformio\packages\framework-arduinoespressif32\tools\sdk\include\config -Ic:\...\.platformio\packages\framework-arduinoespressif32\tools\sdk\include\bluedroid -Ic:\...\.platformio\packages\framework-arduinoespressif32\tools\sdk\include\app_trace -Ic:\...\.platformio\packages\framework-arduinoespressif32\tools\sdk\include\app_update -Ic:\...\.platformio\packages\framework-arduinoespressif32\tools\sdk\include\bootloader_support -Ic:\...\.platformio\packages\framework-arduinoespressif32\tools\sdk\include\bt -Ic:\...\.platformio\packages\framework-arduinoespressif32\tools\sdk\include\driver -Ic:\...\.platformio\packages\framework-arduinoespressif32\tools\sdk\include\esp32 -Ic:\...\.platformio\packages\framework-arduinoespressif32\tools\sdk\include\esp_adc_cal -Ic:\...\.platformio\packages\framework-arduinoespressif32\tools\sdk\include\ethernet -Ic:\...\.platformio\packages\framework-arduinoespressif32\tools\sdk\include\fatfs -Ic:\...\.platformio\packages\framework-arduinoespressif32\tools\sdk\include\freertos -Ic:\...\.platformio\packages\framework-arduinoespressif32\tools\sdk\include\heap -Ic:\...\.platformio\packages\framework-arduinoespressif32\tools\sdk\include\jsmn -Ic:\...\.platformio\packages\framework-arduinoespressif32\tools\sdk\include\log -Ic:\...\.platformio\packages\framework-arduinoespressif32\tools\sdk\include\mdns -Ic:\...\.platformio\packages\framework-arduinoespressif32\tools\sdk\include\mbedtls -Ic:\...\.platformio\packages\framework-arduinoespressif32\tools\sdk\include\mbedtls_port -Ic:\...\.platformio\packages\framework-arduinoespressif32\tools\sdk\include\newlib -Ic:\...\.platformio\packages\framework-arduinoespressif32\tools\sdk\include\nvs_flash -Ic:\...\.platformio\packages\framework-arduinoespressif32\tools\sdk\include\openssl -Ic:\... [... cut ...] -llwip -lnet80211 -lpthread -ljson -lstdc++ -Wl,--end-group
xtensa-esp32-elf-size -B -d .pioenvs\esp32dev\firmware.elf
"c:\python27\python.exe" "c:\...\.platformio\packages\tool-esptoolpy\esptool.py" --chip esp32 elf2image --flash_mode dio --flash_freq 40m --flash_size 4MB -o .pioenvs\esp32dev\firmware.bin .pioenvs\esp32dev\firmware.elf
text	   data	    bss	    dec	    hex	filename
489690	 160552	  32648	 682890	  a6b8a	.pioenvs\esp32dev\firmware.elf
esptool.py v2.1
 [SUCCESS] Took 19.09 seconds

closed 2018-09-01 by smartinick