smartinick/esp32_xiaomi_m365 #13: How to use "telnet"

S-m-i-t-h, 2018-08-28:
Hi,

I get the IP of the ESP32 onces it is connected to wifi network.

Ping to that IP is working fine.

But I execute "telnet 192.168.0.160 36523" on cmd on windows, but I can't establish the connection.

Could you explain please how use telnet in your proyect? Thanks!

smartinick, 2018-08-28:
did you enable developermode and usetelnetserver?
so, you have "#define developermode" active and under the #ifdef developermode, the #define usetelnetserver is active?
by activ i mean not commented out?

if so - and it does not work, let me give it a try...

S-m-i-t-h, 2018-08-28:
OK, I didn't enable developermode.
Now it works fine.
Thank you very much!

closed 2018-08-29 by S-m-i-t-h