smartinick/esp32_xiaomi_m365 #21: Use esp32 as replacement head unit/controller

duncan999007, 2018-12-05:
Would it be possible to fully replace the head unit board with the esp32? Could this firmware be modified to send commands as well, if I wanted to connect a custom throttle or use the motor controller in a different application?

smartinick, 2018-12-05:
yes, yes

closed 2018-12-05 by smartinick

duncan999007, 2018-12-08:
I saw on another post that you've used the ESP32 without the head unit to control the motor. I've been trying for a week to get this to work on my scooter without the BLE unit. Could you share the firmware you got this to work on?

smartinick, 2018-12-09:
> I saw on another post that you've used the ESP32 without the head unit to control the motor. I've been trying for a week to get this to work on my scooter without the BLE unit. Could you share the firmware you got this to work on?

you can look here: [https://github.com/camcamfresh/Xiaomi-M365-BLE-Controller-Replacement/blob/master/BasicMotorControl](url)

that's where i started. though that code is blocking and so on, but with some small modifications it works.  and off course on the esp32 the uart/adc will be different. (i had to change the 10ms delay to 15ms and the min-value from 26 to 28).

but to also give you a answer: no, i'll not share that code (to use esp as ble-module replacement). sometime in the future i'll add a webserver to setup gpio pins, then i can provide a encrypted binary you can load and setup the functionality/gpio pins via webinterface.
as there's other people using my code to run closed-source projects or to sell displays, i don't want to feed those trolls any longer :D

duncan999007, 2018-12-10:
I gotcha, thanks man. I'm just getting started with more sophisticated MCU things